adafruit/Adafruit_LIS3DH #6: Library doesn't compile with ATTiny85

Glench, 2016-07-26:
Trying to get the LIS3DH sensor working with an ATTiny85, the library does not compile. Here is the error I get:

```
Arduino: 1.6.9 (Mac OS X), Board: "ATtiny25/45/85, ATtiny85, Internal 8 MHz"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/glen/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/glen/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/glen/Documents/Arduino/libraries" -fqbn=attiny:avr:ATtinyX5:cpu=attiny85,clock=internal8 -ide-version=10609 -build-path "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/glen/tmp/sketch_jul25c/sketch_jul25c.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/glen/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/glen/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/glen/Documents/Arduino/libraries" -fqbn=attiny:avr:ATtinyX5:cpu=attiny85,clock=internal8 -ide-version=10609 -build-path "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/glen/tmp/sketch_jul25c/sketch_jul25c.ino"
WARNING: Spurious .github folder in 'Adafruit LIS3DH' library
Build options changed, rebuilding all
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH/Adafruit_LIS3DH.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src/Wire.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src/utility/twi.c" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/dev/null"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/glen/Library/Arduino15/packages/attiny/hardware/avr/1.0.2/variants/tiny8" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_LIS3DH" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/src" "-I/Users/glen/Documents/Arduino/libraries/Adafruit_Sensor" "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/sketch/sketch_jul25c.ino.cpp" -o "/var/folders/g5/_p40z6fj53z45x1xvgvxc3g80000gn/T/buildfb32aecad0cc70c77d8b1590509dca12.tmp/preproc/c [... cut ...]
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

1. Try to compile this sketch with the following settings:

```
#include <Adafruit_LIS3DH.h>
void setup() {
}
void loop() {
}
```
[image]
(This screenshot is an attempt to show the core files we're using, the board settings, and the version number of the Adafruit LIS3DH library we're using.)

It seems this error is caused by including Wire.h and/or SPI.h. Any ideas about why this library won't compile for this chip?

(cc @damellis)

ladyada, 2016-07-26:
if you use Adafruit AVR Boards https://github.com/adafruit/arduino-board-index/, our Trinket definition also replaces Wire - but either way its not guaranteed to work. may not fit - we simply don't guarantee it

closed 2016-07-26 by ladyada

Glench, 2016-07-26:
I'm a little confused by your comment, especially given this part of the library's README, which indicates that this library *does* work with the ATTiny85:

[image]

Can you help me clear up the confusion or point me to a resource that would help? I don't understand what Trinket definition you are referring to, or how I would apply that information to my situation.

ladyada, 2016-07-26:
yes with Trinket - this library was (at some point) tested and compiles for adafruit customers who have purchased a Trinket. if this is your first time working with arduino, i strongly recommend against trying to use this lib with '85. its challenging and difficult to debug

Glench, 2016-07-27:
Ah, so there was either custom software used or custom hardware built onto the Trinket that allowed it to work with this library. So it seems like we'll have to rewrite this library using something the ATTiny can understand. Thanks for the clarification!

Kylir, 2016-07-27:
Hi @Glench 
I'm not at all an expert in C or C++, I worked with the Adafruit LIS3DH on an Espruino using Javascript...
But still, I had a quick look at your error log and your have a lot of undeclared constants: SDA, SCL and so on... Do you have a missing dependency, like a `.h` or even a missing `cpp` file somewhere?