adafruit/Adafruit_LIS3DH #14: LIS3DH_DATARATE_LOWPOWER_5KHZ mode is not setting the accelerometer in 5kHz low power mode but in 1.344 kHz High resolution mode

nalexopo, 2018-06-01:
lis.setDataRate(LIS3DH_DATARATE_LOWPOWER_5KHZ);

 Above code is not setting the accelerometer in 5kHz mode but in 1.344 kHz High resolution mode. I read the LIS3DH_REG_CTRL1 register and i get a value b'1001 0111. It should be  b'1001 1111 for 5kHz low power mode and the HR bit in LIS3DH_REG_CTRL5 should be 0. Read Lis3dh datasheet in page 35/54.

erichiggins, 2018-08-03:
@ladyada Is this repo still being maintained? There are several open PRs and Issues, like this one, which have not had responses.

nalexopo, 2018-09-11:
@erichiggins  

 Some suggestions from my part as I just saw that you replied.

1. Don't use the datasheet from adafruit site but rather the one from ST site because it has been revised a few times from the 2010 version your link leads to. This is the 2016 version you can see some differences in the ODR that's why i am talking about 1.344 kHz and not 1250 kHz.

Here the December 2016 version which is the current version. 
[https://www.st.com/resource/en/datasheet/lis3dh.pdf](url)

2. If you are planning on going full speed on this sensor have a look on the application note page 9/59
you may be limited by bandwidth limitations when using HR mode.

https://www.st.com/content/ccc/resource/technical/document/application_note/77/ed/e7/e1/28/5a/45/d6/CD00290365.pdf/files/CD00290365.pdf/jcr:content/translations/en.CD00290365.pdf

T-Mosher, 2024-06-11:
Note, same issue raised via this forum thread:
https://forums.adafruit.com/viewtopic.php?t=211362

Is this issue pending a resolution, stuck, or merely forgotten?

tyeth, 2024-06-17:
Hardware arriving on Wednesday, so I'll have a look + test then

tyeth, 2024-07-01:
Hey @nalexopo @erichiggins @hakeemta @T-Mosher, there is a PR in to add the functionality, see #50 , it would be really appreciated if you folks could give it a thorough testing!

T-Mosher, 2024-07-01:
Sorry, can't test it, don't have the hardware.

tyeth, 2024-07-08:
@nalexopo @erichiggins @hakeemta final testing reminder prod ☝️  before I merge it tomorrow and release anyway. We can always fix it afterwards.

erichiggins, 2024-07-08:
I'm currently traveling and unable to test this

tyeth, 2024-07-08:
Thanks for the thought, and happy travels!

closed 2024-07-16 by tyeth

tyeth, 2024-07-16:
Please feel free to reopen or create a new issue if there are any problems with the release.

commit a07a936a (2024-06-28): Cleanup formatting + swap python comments to cpp 🤫

Resolves issue #14 by adding additional operational modes and adjusting read method.
Original issue "LIS3DH_DATARATE_LOWPOWER_5KHZ mode is not setting the accelerometer in 5kHz low power mode but in 1.344 kHz High resolution mode "
Fixes #14