adafruit/Adafruit_LIS3DH #18: Setting resolution & BDU on incorrect register?

erichiggins, 2019-10-16:
- Arduino board:  **AdaFruit Feather M0 BLE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.10**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **N/A**

I want to set the high-res bit, which is defined on [`LIS3DH_REG_CTRL3`](https://github.com/adafruit/Adafruit_LIS3DH/blob/master/Adafruit_LIS3DH.h#L124-L135). However, it looks like it's being incorrectly set to `LIS3DH_REG_CTRL4` in [`Adafruit_LIS3DH.cpp`](https://github.com/adafruit/Adafruit_LIS3DH/blob/master/Adafruit_LIS3DH.cpp#L137-L138)

Since `writeRegister8` is a private method, I cannot override this without modifying the library outside of my own code (not a best-practice).

**I'm happy to submit a PR to fix this**, but would first like a maintainer to confirm the issue.

Thanks!

ladyada, 2019-10-16:
hi where do you see its in CTRL3?

[image]

erichiggins, 2019-10-16:
Oh shoot, I'm sorry. I was reading from the code in your `.cpp` file and misread the order/labels. 
Thanks for the quick reply and sorry for the noise!

closed 2019-10-16 by erichiggins