adafruit/Adafruit_LIS3DH #29: FIFO mode is not available

matthewfcarlson, 2020-05-12:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  Adafruit Huzzah ESP32

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.12

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

From the docs:
```
The LIS3DH embeds a 32-level FIFO for each of the three output channels, X, Y and Z.
This allows consistent power saving for the system, since the host processor does not need
to continuously poll data from the sensor, but it can wake up only when needed and burst
the significant data out from the FIFO.
In order to enable the FIFO buffer, the FIFO_EN bit in CTRL_REG5 (24h) must be set to ‘1’.
This buffer can work according to the following different modes: Bypass mode, FIFO mode,
Stream mode and Stream-to-FIFO mode. Each mode is selected by the FM [1:0] bits in
FIFO_CTRL_REG (2Eh). Programmable FIFO watermark level, FIFO empty or FIFO
overrun events can be enabled to generate dedicated interrupts on the INT1 pin
(configuration through CTRL_REG3 (22h)).
In the FIFO_SRC_REG (2Fh) register the EMPTY bit is equal to ‘1’ when all FIFO samples
are ready and FIFO is empty.
In the FIFO_SRC_REG (2Fh) register the WTM bit goes to ‘1’ if new data is written in the
buffer and FIFO_SRC_REG (2Fh) (FSS [4:0]) is greater than or equal to FIFO_CTRL_REG
(2Eh) (FTH [4:0]). FIFO_SRC_REG (2Fh) (WTM) goes to ‘0’ if reading an X, Y, Z data slot
from FIFO and FIFO_SRC_REG (2Fh) (FSS [4:0]) is less than or equal to
FIFO_CTRL_REG (2Eh) (FTH [4:0]).
In the FIFO_SRC_REG (2Fh) register the OVRN_FIFO bit is equal to ‘1’ if the FIFO slot is
overwritten.
```

Would y'all be open to a PR?

ladyada, 2020-05-12:
yep we welcome PRs :)

siddacious, 2020-05-12:
yes please :)

matthewfcarlson, 2020-05-13:
Okay- I have it working and added an example. I'll clean it up and post a PR tomorrow ish?