adafruit/Adafruit_LIS3DH #32: Interrupt pin always HIGH after version 1.1.3 for Attinyxy4

JasXSL, 2020-05-31:
After the update was made to use BusIO, the interrupts stopped working for the new ATTINY series. 

The interrupt pin will always be HIGH, regardless of if it's moving or standing still.
Version 1.1.2 doesn't have this issue.

Using this sketch which runs some neopixels whenever an interrupt is received: https://gist.github.com/JasXSL/ed0f35498f64f8736e2fa730bde88b75

On 1.1.2 the pixels are off until a shake is detected (expected behavior), on later versions the pixels are always on.

Using:

- Adafruit_LIS3DH library cloned from github an hour ago.
- Adafruit_BusIO library cloned from github including the fixes related to this issue https://github.com/adafruit/Adafruit_BusIO/issues/29
- Arduino 1.8.12
- attiny1604 with core: https://github.com/SpenceKonde/megaTinyCore
- These are my settings:
[image]

siddacious, 2020-06-20:
As it turns out, this is because the old `setClick` method was blowing away the DRDY interrupt which is on by default. I'll add a accessors to enable/disable DRDY to allow for similar-enough behavior.

siddacious, 2020-06-23:
@JasXSL I just released version 1.2.0 that adds the `enableDRDY` call to enable or disable the Data Ready interrupt that is causing the INT pin to always seem high. Once you update your `Adafruit_LIS3DH` to 1.2.0, you can then add `lis.enableDRDY(false);` before your call to `setClick` to disable the data ready interrupt so that only tap events will raise the INT pin.

 You can see [the documentation](http://adafruit.github.io/Adafruit_LIS3DH/html/class_adafruit___l_i_s3_d_h.html#acca8da6249158d39c9fd6c799df94e6e) for more details.

closed 2020-06-23 by siddacious