adafruit/Adafruit_LIS3DH #40: Docs not updating

dherrada, 2021-04-08:
```
Run bash ci/doxy_gen_and_deploy.sh
Setting up the script...
Cloning into 'Adafruit_LIS3DH'...
Generating Doxygen code documentation...
Grabbing default Doxyfile
This is a Commit, Uploading documentation...
Uploading documentation to the gh-pages branch...
Adding all files
Changes to commit
Git committing
[gh-pages ec835a5] Deploy docs to GitHub Pages from commit ce859eafe7
 1 file changed, 1 insertion(+), 1 deletion(-)
Git pushing
Error: Process completed with exit code 128.
```

ladyada, 2021-04-08:
yeah the GH repo secret got deprecated, so we need to remove it from every repo that had it
go to
https://github.com/adafruit/Adafruit_LIS3DH/settings/secrets/actions
delete the GH_REPO_TOKEN so it uses the org token, then kick CI!

dherrada, 2021-04-08:
ah ok. Do you think you could message that to me? I don't think I have it

closed 2021-04-08 by dherrada