adafruit/Adafruit_LIS3DH #47: Scaled readings in 16G range to high

marc-gross, 2022-06-26:
Hello,
I think I found a bug in the .cpp at line 219, shouldn‘t the lsb_value for 16G be set to 32?

ladyada, 2022-06-28:
the datasheet says otherwise. if you have evidence for this bug (not 'guessing') please post proof!

marc-gross, 2022-06-29:
Hi,
just checked it, you are right about that, sorry.

My assumption was based on the experience that I got readings above 200m/s^2 @16g range, what is way over spec. So I was looking for the reason and found this ‚unlogical‘ stepsize, where the sensitivity values double between 2/4/8g and for 8g to 16g the step is x3. Since i got slightly high gravity readings in the first test too, it seemed obvious to me that there must be some problem with the scaling.
Now I‘m curious, if the sensor would provide the theoretical max 24g output with this sensitivity. Need to improve my waterrocket…

King regards
Marc